ggventurini/python-deltasigma #61: /usr/bin/ld: cannot find -lcblas

snelgar, 2016-10-01:
deltasigma is working nicely for me on a Mac 10.9 setup, but not on 10.11 and not on Ubuntu. The problems are all to do with cblas. 
On Mac OS 10.11 and up /usr/include is "restricted", and opencblas puts things in a weird place (/opt/local/include/cblas.h).
On my linux 4.5.3-x86_64 I have /usr/include/cblas.h, and instead get the message in the title when I "import deltasigma"
Any help appreciated; your python version of deltasigma is much more useful than the Matlab one.

itdaniher, 2016-10-01:
HI @snelgar - can you include what version you're using of deltasigma and the minimum commands to trigger the error on Linux? I'll take a look.

snelgar, 2016-10-01:
Hi Ian:
Just importing is good enough to trigger the problem:

> $ python3
> Python 3.5.2 (default, Sep 10 2016, 08:21:44) 
> [GCC 5.4.0 20160609] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import deltasigma
> /usr/bin/ld: cannot find -lcblas
> collect2: error: ld returned 1 exit status
> /usr/bin/ld: cannot find -lcblas
> collect2: error: ld returned 1 exit status

and here’s the version number
> >>> deltasigma.__version__
> ‘0.2.2'

I installed via “pip3 install deltasigma”, if that’s relevant.

	thanks
		martin snelgrove

snelgar, 2016-10-01:
Hi again Ian:

FYI, on Mac OS 10.11 (“El Capitan”), which is the rootless thing that doesn’t allow installations into /usr/include, the bad news looks like this:

> $ python3
> Python 3.5.2 (default, Sep 28 2016, 18:08:09) 
> [GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.38)] on darwin
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import deltasigma
> /var/folders/l3/lf01jzh90mg0f5dc_mdd4nl00000gp/T/tmprk8tjx8r/source.c:1:10: fatal error: 'cblas.h' file not found
> #include <cblas.h>
>          ^
> 1 error generated.
> /var/folders/l3/lf01jzh90mg0f5dc_mdd4nl00000gp/T/tmprk8tjx8r/source.c:1:10: fatal error: 'cblas.h' file not found
> #include <cblas.h>
>          ^
> 1 error generated.
> /usr/local/lib/python3.5/site-packages/deltasigma/_config.py:54: UserWarning: Cannot find the path for 'cblas.h'. You may set it using the environment variable BLAS_H.
> NOTE: You need to pass the path to the directories were the header files are, not the path to the files.
>   warn("Cannot find the path for 'cblas.h'. You may set it using the environment variable "
> >>> deltasigma.__version__
> '0.2.2'

and a lot of that is coming from “get_info", per

> $ python3
> Python 3.5.2 (default, Sep 28 2016, 18:08:09) 
> [GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.38)] on darwin
> Type "help", "copyright", "credits" or "license" for more information.
> >>> from numpy.distutils.system_info import get_info
> >>> get_info("blas")
> /var/folders/l3/lf01jzh90mg0f5dc_mdd4nl00000gp/T/tmp7v2u8_xn/source.c:1:10: fatal error: 'cblas.h' file not found
> #include <cblas.h>
>          ^
> 1 error generated.
> /var/folders/l3/lf01jzh90mg0f5dc_mdd4nl00000gp/T/tmp7v2u8_xn/source.c:1:10: fatal error: 'cblas.h' file not found
> #include <cblas.h>
>          ^
> 1 error generated.
> {'library_dirs': ['/usr/lib'], 'libraries': ['blas', 'blas']}

I muddled through the code in _config and found it guessing that /usr/include might work. Setting BLAS_H didn’t do much for me either.

	thanks
		martin s.

3mdj23, 2016-10-05:
Hi Ian,

Jut following up on the Linux (Linux d5d56fb00c3f 4.4.22-moby #1 SMP Sun Sep 25 22:54:41 UTC 2016 x86_64 GNU/Linux) import issue that Martin mentioned previously. 

I installed deltasigma using pip. Then, I installed the libblas-dev package from Debian jessie and that solved the cblas.h not found issue. This then gave me the error:
```
root@d5d56fb00c3f:/work/software/algo# python
Python 2.7.11 (default, Apr 21 2016, 17:11:33)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import deltasigma as ds
/usr/local/lib/python2.7/site-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
/usr/bin/ld: cannot find -lcblas
collect2: error: ld returned 1 exit status
/usr/bin/ld: cannot find -lcblas
collect2: error: ld returned 1 exit status
>>>
```

The blas library exists, but not cblas:
```
root@d5d56fb00c3f:/work/software/algo# ls /usr/lib/libcblas*
ls: cannot access /usr/lib/libcblas*: No such file or directory
root@d5d56fb00c3f:/work/software/algo# ls /usr/lib/libblas*
/usr/lib/libblas.a  /usr/lib/libblas.so  /usr/lib/libblas.so.3	/usr/lib/libblas.so.3gf

/usr/lib/libblas:
libblas.a  libblas.so  libblas.so.3  libblas.so.3.0
root@d5d56fb00c3f:/work/software/algo#
```

As a work-around I symlinked libblas.so to libcblas.so:
```
#> apt-get update
#> pip install deltasigma
#> apt-get install libblas-dev
#> cd /usr/lib
#> ln -s libblas.so libcblas.so

root@bf4be19467c9:/usr/lib# python
Python 2.7.11 (default, Apr 21 2016, 17:11:33)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import deltasigma as ds
>>>
```

Not sure if it's relevant but apparently there have been a few name changes in the atlas and/or blas libraries recently (see: https://www.centos.org/forums/viewtopic.php?t=48543).

Let me know if you need more information.

Kindest regards,
Matthew

ma-laforge, 2021-05-07:
Hi @snelgar,

### My own (Julia) port of the ΔΣ library
Just wanted to mention there is a Julia version of Richard's toolbox. I've got a good portion of it ported, and I can probably port other portions if you give me minimum working examples to work from.
-->https://github.com/ma-laforge/RSDeltaSigmaPort.jl

### About Julia
If you don't already know, Julia is the MIT-seeded language that takes what Matlab gave us to a whole new level:
 - https://julialang.org/
 - https://en.wikipedia.org/wiki/Julia_(programming_language)

### Known issues
At the moment, the biggest issue appears to be is in the conversion from ABCD -> zpk||rational.

(Same goes to the others on this thread :))