ggventurini/python-deltasigma #70: PlotExampleSpectrum() fails in Python3 due to changed '/'-operator meaning

ldpgh, 2019-08-20:
`PlotExampleSpectrum(ntf, M=1, osr=osr, f0=f0)`

Traceback (most recent call last):

  File "<ipython-input-83-ad1adad17f22>", line 1, in <module>
    PlotExampleSpectrum(ntf, M=1, osr=osr, f0=f0)

  File ".....\lib\site-packages\deltasigma\_PlotExampleSpectrum.py", line 103, in PlotExampleSpectrum
    plt.plot(freq, dbv(spec0[:N/2 + 1]), 'c', linewidth=1)

TypeError: slice indices must be integers or None or have an __index__ method

---
... running Python 3.7.0
---

Change line  File ".....\lib\site-packages\deltasigma\_PlotExampleSpectrum.py", line 103, in PlotExampleSpectrum
`    plt.plot(freq, dbv(spec0[:N/2 + 1]), 'c', linewidth=1)`
to
`    plt.plot(freq, dbv(spec0[:int(N/2.0) + 1]), 'c', linewidth=1)`

or whatever is convinient to let  `N/2` return an integer number.

ldpgh, 2019-08-20:
same issue on line 106

  File "<ipython-input-88-ad1adad17f22>", line 1, in <module>
    PlotExampleSpectrum(ntf, M=1, osr=osr, f0=f0)

  File ".......\lib\site-packages\deltasigma\_PlotExampleSpectrum.py", line 106, in PlotExampleSpectrum
    plt.plot(freq, dbp(spec_smoothed[:N/2 + 1]), 'b', linewidth=3)

TypeError: slice indices must be integers or None or have an __index__ method

ldpgh, 2019-08-20:
... and on line 87, 88
    f1_bin = np.round(f1*N)
    f2_bin = np.round(f2*N)
have been modified to
    f1_bin = int(np.round(f1*N))
    f2_bin = int(np.round(f2*N))

ldpgh, 2019-08-20:
... and voila got a pretty nice spectrum. 

And not to forget ... Thank you for all the effort you spent.

Yuki-F-HCU, 2020-09-02:
I fixed these issues in my folked repository. Thank you for your report.