ggventurini/python-deltasigma #74: [Announcement] python-deltasigma is forked and maintained now.

Yuki-F-HCU, 2020-09-09:
At first, I apologize for my poor English in advance (I'm Japanese).  

I'm interested in Delta-Sigma modulation and trying to use this library for development.  
But, I think the future of this library is unclear because I haven't checked the contributors of this library have been maintained in these years.  

So, I decided to fork this repository and maintain uniquely in [Yuki-F-HCU/python-deltasigma](https://github.com/Yuki-F-HCU/python-deltasigma).  

I am trying to maintain this repository and passed the build test in Travis CI.  
If you are interested in using this python-deltasigma ongoingly, you are welcome to freely contriubute to its development. Please check out.

ma-laforge, 2021-05-07:
That's great to hear @Y-F-Acoustics!

Were you aware you can turn on the "Issues" tab to help others open issues directly in your own fork?
Go to your own repo's "Settings" tab, go to "`Options|Features`", then:
 - Check the "Issues" box.

(see below)

[image]

Y-F-Acoustics, 2021-05-08:
Dear @ma-laforge,

Thank you for your suggestion.
I applied this setting in my forked repo.

closed 2021-05-18 by Y-F-Acoustics