ggventurini/python-deltasigma #76: synthesizeNTF(order, OSR, opt=2,f0=0.25)

AbdelrahmanHassanMuhammed, 2020-09-30:
I can't synthesize NTF at center freq of 0.25, I get TypeErro 
**TypeError: 'float' object cannot be interpreted as an integer**

Y-F-Acoustics, 2020-10-01:
I folked this library and maintain in [Y-F-Acoustics/python-deltasigma] (https://github.com/Y-F-Acoustics/python-deltasigma).

This may synthesize NTF. I will try to check it can synthesize. Please paste your code which clarifies order and OSR.

AbdelrahmanHassanMuhammed, 2020-10-01:
Here's the code 
`order = 4
OSR = 64

H0 = synthesizeNTF(2,64,2,1.5,0.25)
subplot(121)
plotPZ(H0, markersize=5)
title('NTF Poles and Zeros')
f = np.concatenate((np.linspace(0, 0.75/OSR, 100), np.linspace(0.75/OSR, 0.5, 100)))
z = np.exp(2j*np.pi*f)
magH0 = dbv(evalTF(H0, z))`

[image]

Y-F-Acoustics, 2020-10-01:
Thanks a lot!
This problem is not your fault (It was already reported on #72 ). I checked the bug on this library and fixed them on my forked repository.

If you can install and use my folked python-deltasigma, please check it and pull request.

AbdelrahmanHassanMuhammed, 2020-10-01:
Thanks a lot, it works.

Y-F-Acoustics, 2020-10-01:
In case of Python, np.zeros returns the zeros matrix like MATLAB.
So, input parameters in zeros must be integers. You can write like below,

G = (np.zeros((order//2, )), H[1], 1)
or
G = (np.zeros((np.int(order/2), )), H[1], 1)

You must be warn that the python returns float automatically like MATLAB. 
For example, the formula '1/2' doesn't return '0', but '0.5'.

AbdelrahmanHassanMuhammed, 2020-10-01:
Ok, thanks a lot

closed 2020-10-01 by AbdelrahmanHassanMuhammed

Y-F-Acoustics, 2020-10-01:
Sorry for my poor English.