ggventurini/python-deltasigma #79: Python 3.9.6 -> `AttributeError: module 'fractions' has no attribute 'gcd'`

jolonf, 2021-08-10:
Importing `deltasigma` produces the error:

`AttributeError: module 'fractions' has no attribute 'gcd'`

From: https://docs.python.org/3.5/library/fractions.html#fractions.gcd

"Deprecated since version 3.5: Use math.gcd() instead."

haunma, 2021-11-14:
I have this problem too.  Not sure going with a random fork, as suggested above, is a good idea...

Y-F-Acoustics, 2022-01-07:
I'm sorry for late reaction to this problem.

I maintain this module in [https://github.com/Y-F-Acoustics/python-deltasigma](https://github.com/Y-F-Acoustics/python-deltasigma) but it is no longer maintained for Python 3.5 because it has not been supported officialy.

I apologize for inconvenience but please consider updating the Python version above 3.6.

Best Regards,