ggventurini/python-deltasigma #84: AttributeError: module 'numpy' has no attribute 'float'.

olisnr, 2024-06-02:
hi 
when i try to use this interesting module with python 3.10 there popup many error messages

the first is: `Exception has occurred: AttributeError
module 'numpy' has no attribute 'float'.
    
its possible to change the float to float64, but then the next error comes. the function `scipy.signal.step2()` was replaced with `step()`. after `import collections` it seems to need a `collections.Iterable = collections.abc.Iterable`. the function `gcd()` is now in the module `math` instead of `fractions` and there are more problems i didnt found out by now.

thus my question: is `deltasigma` already ported to a more modern python version? and if Yes, where to find it? or should i try to fix all errors my self and somehow upload it (if im lucky)?

olisnr, 2024-06-05:
i got a part running on python 3.10
here is my diff:
[ds.diff.zip](https://github.com/user-attachments/files/15596852/ds.diff.zip)

DanielGlouseer, 2024-10-10:
I just faced the same condition with you and couldn't figure it out...

olisnr, 2024-10-10:
@DanielGlouseer does the diff work for You?

DanielGlouseer, 2024-10-12:
I encountered another error. First is: 'Exception has occurred: AttributeError module 'numpy' has no attribute 'float' . Then I **_change to float64_** and it seems ok. Also I ignore anywhere mentioned ''**_plt.ishold_**''.  The next error I faced is ''Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure''.  I tried use code as **_matplotlib.use('TkAgg')_**, now I could get the correct picture.

olisnr, 2024-10-22:
can You make a diff of all Your work?