ggventurini/python-deltasigma #85: Maybe use newer fork

leonhardklein, 2024-09-09:
Hi,
I hope it is not inpolite to open a issue like this:
It took me some time until I confirmed my python skills for updating outdated code are not as good as the fork in 

https://github.com/hpretl/python-deltasigma

Since this repo seems to be quite dead, the fork may save other people some time - so I hope it is ok to highlight this in here. (there are other forks as well, but that one seemed to be the most advanced as today).

Best Regards,
leo

fporcari, 2024-09-10:
> Hi,
> I hope it is not inpolite to open a issue like this:
> It took me some time until I confirmed my python skills for updating outdated code are not as good as the fork in [...]
Hi 
I'm a good friend of ggventurini. He passed away about 6 years ago. 
His mother gave me the access to the project but this project is not in my area of interest.
I think your fork could be a good solution for all deltasigma users.
Best Regards
Francesco

leonhardklein, 2024-09-10:
oh.. That's very sad to hear!
I'm not the owner of any other repository, but as this repo is the top result of most search engines (and http://www.python-deltasigma.io/ is linking to it) -- could you put a warning on the first page? Maybe just mention people should search for an updated version to not provide a link to another outdated/malicious software?

olisnr, 2024-10-11:
@leonhardklein thanks for the link!